sneakytreesnake/StealthOrbiter #2: filament_switch_sensor toolhead_sensor question

suizhenmin, 2023-01-20:
Hi orBiter Filament Sensor sensor releases V2.2 version. Now no matter how to set it, the filament state is displayed. How can I make it?

The following is my setting

[filament_switch_sensor toolhead_sensor]            
pause_on_runout: False
#runout_gcode: PAUSE #PAUSE WHEN FILAMENT RUNS OUT - DOES THIS SCREW WITH ERCF?
# filament sensor wired to the ERB or Printer MCU
#switch_pin: ercf:gpio25 #ERB
switch_pin: !PA1 #Printer MCU - FYSETC CHEETAH_MIX
#switch_pin: ^P1.27 #Printer MCU - BTT SKR board

sneakytreesnake, 2023-01-20:
It should just require inverting the pin in klipper config.

I believe the orbiter sensor page has more instructions on how to set the sensor software up.

suizhenmin, 2023-01-21:
Does the original OrbiterSensor.cfg configuration need to be disabled?

sneakytreesnake, 2023-01-21:
Yes disable the original .cfg file, and use the new OrbiterSensorV2.2.cfg file.

suizhenmin, 2023-01-22:
I almost gave up, now when the _ERCF_CALIBRATE is calibrated, the filament reaches the entrance of the extruder, but the extruder is in the state of returning material, maybe I will find an official Orbiter v2.0 base file to use ercf plan.

suizhenmin, 2023-01-22:
I checked your STL file, you used a material break detection alone, installed under the extruder base, can you provide the following BOM list?

sneakytreesnake, 2023-02-06:
I've added a BOM and updated the description to make the installation more clear.

closed 2023-02-06 by sneakytreesnake