sneakytreesnake/StealthOrbiter #4: Is there a mounting plate for ender3 v2 X carriage?

Gil80, 2023-02-24:
Hi!
This looks very good and I'd like to print the parts but I'm not too sure which parts and where to source them.
I use the Orbiter V2 with the Orbiter filament sensor and Phaetus Draghon HF hotend.
My ender 3 v2 X carriage is using wheels/rollers, not linear rails with 4010 heatbreak fan and 5015 radial fan.

Can you please help me to figure out which parts to print?

sneakytreesnake, 2023-02-26:
Hello, this mod uses the standard stealthburner carriage. I haven't done this myself, but if you manage to find an adapter plate (or design one) that adapts the stealthburner carriage to the ender 3 that would work.

A quick google search revealed this:

https://www.thingiverse.com/thing:5191205

and you should be able to print the rest from the Voron stealthburner Github repo

sneakytreesnake, 2023-02-26:
Closing as issue is not related to this mod.

closed 2023-02-26 by sneakytreesnake