sneakytreesnake/StealthOrbiter #7: Install Guide

marc-gist, 2023-03-26:
Thank you for sharing! Do you have a link to any install guides/videos... I'm not really sure about how one installs on to the motor the chain guide, etc, or what exactly all the parts are that are needed to be printed between your stl and the linked one.

sneakytreesnake, 2023-03-26:
Give me a few days and I'll put together an install guide/video 😀

sneakytreesnake, 2023-04-01:
Hello, I've added a set of diagrams and descriptions. I've kept the links to the other design, as I don't yet have permission to upload them as part of this design.

Let me know how you get on, and if any other info would be useful!

sneakytreesnake, 2023-04-10:
Closed as install guide has been added, and no further comments made

closed 2023-04-10 by sneakytreesnake