sneakytreesnake/StealthOrbiter #8: [REQ] Filament sensor version with PG7 gland

ndng28, 2023-04-03:
Hello, Thanks for working on this mod. 

Is there a version of the mount with both the filament sensor and PG7 cable gland?

sneakytreesnake, 2023-04-04:
Hello,

No there isn't, but I'm happy to upload one. Expect it in the next day or so.

If you are trying to use the filament sensor with ERCF, I'd recommend using the sensorless branch. I am using it and find that it works flawlessly. You can read about it [here.](https://github.com/moggieuk/ERCF-Software-V3)

ndng28, 2023-04-04:
Thanks! I have read through discord and the post you linked at nauseam. From what I gathered was that it was more reliable to have a micro switch...I understand sensorless works really well but has to be setup "right" or else it wont work. I have already wasted time with a hall-sensor so was hoping to go the micro-switch route rather than sensorless.

Either way, I would super appreciate if you end up creating one with both (fila-sensor and PG7). Let me know if you have patreon or something I else where i can buy you a coffee :)

Cheers

sneakytreesnake, 2023-04-04:
I've added a PG7 filament sensor mount. Keep the print orientation the same (it's important the ball bearing runs smoothly in the hole). You will need to use supports.

Cheers for the offer but I am just glad other people find my work useful :-)