sneakytreesnake/StealthOrbiter #13: PG7 mount and ADXL345

garyd9, 2023-05-28:
Your suggested mounting for a Can bus toolhead using the PG7 mount has the CANBUS board mounted at an diagonal angle.  (here: https://github.com/sneakytreesnake/StealthOrbiter/tree/main/STL#canbus-toolhead-ebb36-sht36-pg7-straingland) This will likely result in the built ADXL345 becoming useless (because it can't differentiate which axis is which.)

It might be better to suggest using one of the MANY available CAN adapter mounts to ensure the adapter is mounted square with the toolhead.  For example: https://github.com/KayosMaker/CANboard_Mounts/blob/main/STLs/BTT_EBB36_1.0_Mounts/BTT_EBB36_1.0_mounts_plain/EBB36%201.0%20orbiter%202.0%20mount.stl

The adapter can even be mirrored so the adapter board is oriented with the CAN wiring going down (instead of up) to make wiring to the PG7 easier.

sneakytreesnake, 2023-05-28:
I don't believe that having the ADXL345 on a diagonal would cause any issues, input shaper takes the sum of the accelerations to determine the resonances. That's why you don't need to explicitly define or align the ADXL axis in klipper.

garyd9, 2023-05-29:
Perhaps you're right.  I'll have to give it a shot and see.  :)  Thank you for the model.  Trying to get things going with it now, but having a slight issue with the SB body.  I'll open another issue for that.

closed 2023-05-29 by garyd9